dxoverdy/Alpha2MQTT #1: ESP32 support

mercurystorm, 2022-09-13:
I love the idea of this and looking to build my own - Do you know if this would work on an ESP32 instead of an ESP8266?

Benefit being that the ESP32 is much more powerful

dxoverdy, 2022-09-13:
This is a Wemos D1 ESP8266 vs D1 ESP32 circuit layout.
[image]

Instead of adding
http://arduino.esp8266.com/stable/package_esp8266com_index.json
to your Arduino Board Manager URLs, you'd add
https://dl.espressif.com/dl/package_esp32_index.json

And in Board Manager, search out esp32 to install appropriate libraries for the ESP32 Dev module, WEMOS LoLin32 or WEMOS D1 MINI ESP32.

To get the code to compile will involve changing WiFi8266.h to WiFi.h and SoftwareSerial.h to HardwareSerial.h and utilising pins such as IO18, IO19 and IO23 instead of D5, D6 and D7 respectively, seeing as the ESP32 has more than one hardware based serial port.

Let me source one, I'll tweak the code and either create a fork or see if there is a way to #define it to death.  Unless you fancy being a remote tester with my attempts? :)

mercurystorm, 2022-09-13:
Happy to be a remote tester, I already have an ESP32 lying around in the draw, and have ordered the other parts off amazon, but they will take a day or 3 to arrive (Ireland)

mercurystorm, 2023-04-11:
Just as an update on this - I finally ordered a MAX3485 and decided to play around with the ESP32 to see if I could get it working - If anyone else plans to use an ESP32 these are the changes I made

**Alpha2MQTT.ino**
Changed Line 19
`#include <ESP8266WiFi.h>`

to 

`#include <WiFi.h>`

************************************

**Definitions.h**
Added Line 13
`#define LED_BUILTIN 2`

************************************

**RS485Handler.h**
Changed the following pin assignments

`#define SERIAL_COMMUNICATION_CONTROL_PIN D5	// Transmission set pin
#define RX_PIN D6							// Serial Receive pin
#define TX_PIN D7							// Serial Transmit pin`

to

`#define SERIAL_COMMUNICATION_CONTROL_PIN 18	// Transmission set pin
#define RX_PIN 19							// Serial Receive pin
#define TX_PIN 23							// Serial Transmit pin`

Added Package: **ESPSoftwareSerial**

dan-s-github, 2023-06-11:
I ordered a m5stack ATOM RS485 Kit https://shop.m5stack.com/products/atom-rs485-kit and also an atom s3 that includes the display and should look pretty neat and doesn't require an additional power supply.

will try to build and install using the comments above

https://docs.m5stack.com/en/atom/atomic485

RX_PIN 22
TX_PIN 19

dxoverdy, 2023-07-23:
ESP32 support now added.  Please see instructions in 1.21 and 1.22.  With regards, Dan.

closed 2023-07-23 by dxoverdy