dxoverdy/Alpha2MQTT #2: Ground?

jt1971, 2022-09-27:
Good morning.  I was delighted to find this having recently taken delivery of a Smile5 inverter.
Just waiting for all the components to arrive to set up as per your excellent notes.
Can I just confirm from the diagram you don't have a ground connection specifically to the RJ45 connector? Not sure which are the ground wires used by the Alpha and my electronics knowledge is not as advanced!

Thanks.

dxoverdy, 2022-09-27:
Ground (as in next to A+/B-) on the MAX3845 isn’t required.  It’s if you
have outer shielding on your cable as opposed to leveraging twisted pair
which is inherently shielded via the twists.

As the Smile is an RJ45 with twisted pair it’s shielded already, and likely
no pin on the Alpha is a ground anyway.

jt1971, 2022-10-02:
Thanks.

Al installed and working perfectly on my Smile5 (just waiting batteries).  Can I congratulate you on really clear and helpful instructions, both the code and the wiring. Can I get you a virtual coffee?!

dxoverdy, 2022-10-03:
That's really kind, and I am really glad you went ahead and built it and had success!  I didn't have anything set up by way of ko-fi, it wasn't the main driving force...  But, I have done so and set up the sponsor link to ko-fi on the right hand side of the Alpha2MQTT repository homepage :)  Thanks again.

If you have any pictures of your build in situ with your inverter I've love to see them and put on the gallery.

Finally, could I ask which pins ended up being A+/B- on the Smile 5 for the documentation?

jt1971, 2022-10-18:
Hi there,

Apologies for the delay, but have been waiting for the batteries to be installed before I sent though a picture.

Thanks again,

Justin
￼￼

JohnnyZeta, 2022-11-04:
Hi jt1971! I also have a SMILE5 inverter, and I'm struggling to find a ModBus response from the RJ45 plugs on the device.
I tried every couple of wire (white blue/blue, white green/green and so on) on every RJ45 port that seems possible (I tried INV, LAN, CAN), but still no response.

My model is [this one](https://alpha-ess.co.nz/wp-content/uploads/2021/02/Installation-Manual-Storion-Smile-5-V2.pdf).

How do you connected yours?

Thank You in advance!

danielyoung1000, 2022-11-04:
On Page 51 of your manual there, ensure 'Modbus' is enabled on your device.  In the modbus documentation (pic below) it also says "LAN2 or CAN - Modbus must be activated in the display"
[image]

In effect, it's the most right hand port.

I believe it is the same two pins as my B3 - blue white and blue.  Blue white to +.

dxoverdy, 2022-11-04:
Bah, was signed in with my work account.  Let me know how you get on.

dxoverdy, 2022-11-04:
Any pictures of your working device would be gratefully received.  I'd love to see other people's efforts!

ffrog8, 2022-11-11:
@JohnnyZeta I think I'm having similar issues with my smile5. Could you confirm you inverted RX-TX pair on the D1 - do you mean you switched the D1 mini's pins D6 and D7?

dxoverdy, 2022-11-11:
I suspect @JohnnyZeta put D6 to Rx and D7 to Tx, instead of the other way round.

ffrog8, 2022-11-11:
Ahhh okay. @JohnnyZeta did you have to enable modbus in your inverter settings? I don't even seem to have that setting. Thanks.

JohnnyZeta, 2022-11-11:
Hi @ffrog8!
Yes, I inverted D6 and D7, so that was my error.

I enabled slave modbus mode in the inverter's settings; you should found it under Menu -> Setting (1111 default password) -> Function -> Modbus. I also restarted the inverter, but I don't know if is needed/relevant.

ffrog8, 2022-11-11:
Cheers mate. I dunno if its a firmware difference or set up difference but i have no Modbus (or parallel) option in my Function sub menu. I'm running EMS 1.02.41, BMS 1.02 and Invert version 01.97 - how does that compare to your versions? Set up wise I have the smile 5 with 6.6kw of panels and the 13kwh battery - hybrid i think they call it.

closed 2022-11-22 by jt1971