dxoverdy/Alpha2MQTT #3: Which register for Alpha5 PV array

jt1971, 2022-11-22:
Hi. I know this will seem a bizarre question, but I cannot actually find the register that relates to the two DC connected PV arrays on my Smile5.  

REG_PV_METER_R_TOTAL_ACTIVE_POWER_1 reports values for my AC coupled separate solax inverter through a 2nd ACR10R meter.

The only way which does seem odd, is for me to sum the output of REG_INVERTER_HOME_R_PV1_POWER_1 + REG_INVERTER_HOME_R_PV1_POWER_2 and multiply by 10, which approximates to the instantaneous PV output as displayed on the inverter screen itself.

I'm sure I'm missing something, but having looked through all the registers, I cannot find a register that reflects the PV output from the arrays that are connected directly to the inverter.

Thanks.

SorX14, 2022-11-23:
https://github.com/dxoverdy/Alpha2MQTT/blob/master/Alpha2MQTT/RegisterHandler.cpp#L3102 looks to be dividing by 10 incorrectly, i.e. `REG_INVERTER_HOME_R_PV1_POWER_X` registers are giving values that are 10 times less than what it should be.

Those two registers are for the strings of panels connected to the SMILE5, at least they are for me. They match the readings available from the AlphaESS Cloud dashboard (if you check the XHR requests).

dxoverdy, 2022-11-23:
The registers which consist of a _1 and _2 at the end are in fact the same register but a lower and upper end because the number is potentially so large it needs two registers to store it in its entirety.  In the code I’ve commented out the _2s because reading handled registers (ie what gets spat out in a repeating state) does all the hard work of combining the two in the correct way.  What you added there were the two component parts of one array and multiplied and out of chance it got somewhere near.

You want to expose these in one of your repeating states, presumably the 10 second one:

REG_INVERTER_HOME_R_PV1_POWER_1
REG_INVERTER_HOME_R_PV2_POWER_1
REG_INVERTER_HOME_R_PV3_POWER_1

etc etc.  And add those together to get total DC input.

P.S. what’s the use case?  I understand individuals for plotting charts for east/west arrays for example.

dxoverdy, 2022-11-23:
> https://github.com/dxoverdy/Alpha2MQTT/blob/master/Alpha2MQTT/RegisterHandler.cpp#L3102 looks to be dividing by 10 incorrectly, i.e. `REG_INVERTER_HOME_R_PV1_POWER_X` registers are giving values [...]

Noted and correct.  Will modify and upload later.  The joy of copy and pasting!

dxoverdy, 2022-11-23:
Amended to remove the 0.1 mulitplication.  Let us know how you get on summing up the appropriate number of
REG_INVERTER_HOME_R_PV1_POWER_1
REG_INVERTER_HOME_R_PV2_POWER_1
REG_INVERTER_HOME_R_PV3_POWER_1
and if good, close the issue.

Thanks very much
Dan

jt1971, 2022-11-26:
Perfect.  Only just been able to update and test.  Thanks. PV1+PV2 now exactly mirrors the Smile5 display.

closed 2022-11-26 by jt1971