dxoverdy/Alpha2MQTT #8: Set charge / discharge times?

Spookster, 2023-05-15:
*Havent installed yet
Before I dive in, is there a documented way to set discharge / charge times from the interface via code example? (Home Assistant)
Im running a SMILE5 and the documentation and HA Dashboards youve provided doesnt seem to cover that specifically (although it would be possible)

Does anyone have a HA example? or did I miss it?

[image]

dxoverdy, 2023-05-15:
The registers here (
https://github.com/dxoverdy/Alpha2MQTT/blob/4966f1f971f7648a23bdbb75225ca94c7a37ab1e/Alpha2MQTT/Definitions.h#L398)
write those settings.

For example, writing 1 to  REG_TIMING_RW_TIME_CHARGE_START_TIME_1
Will set a start charge time of 00:01.

Alternatively, you can set charge or discharge via an mqtt message.

Dan

Spookster, 2023-05-15:
So theres nothing in the Alpha Write Register HA pages that specifically set it? (ie no dashboard pages that people have created specifically for that control?)
Ill delve in deeper once its running
Thanks Dan, your work seems like just what I want.. let the fun begin.....

Spookster, 2023-05-16:
So this would give a time of 0.01?

```
service: mqtt.publish
data:
  retain: false
  topic: Alpha2MQTT/request/write/register/raw/data
  payload:
    registerAddress: "0x0851"
    value: 1
    dataBytes: 2
  qos: "0"
```
And this 2300?
```
service: mqtt.publish
data:
  retain: false
  topic: Alpha2MQTT/request/write/register/raw/data
  payload:
    registerAddress: "0x0851"
    value: 2300
    dataBytes: 2
  qos: "0"
```

dxoverdy, 2023-05-16:
There is a dashboard which lets you specify a register address and a value
and it will write it.  It’s called Alpha Inverter Control or something like
that.

If you read the docs I’m sure the charge discharge times are set as
minutes, so 0 would be midnight, 1439 would be 23:59 (60*23)+59.  Although
they were the first registers I played with when it comes to writing I just
can’t remember.  What I’d do is write 0 and see what happens, then 10, then
60, etc….

Regarding payloads, download mqttexplorer and do your initial testing
through that.  It’s an invaluable tool.

What is your intention here can I ask?

Dan

Spookster, 2023-05-16:
Nothing too specific I guess, currently using Charles Gillanders code to switch to mains power to run a heat pump when temperatures drop early morning to preserve the battery for use during the day. 
Normally the battery is set to discharge mode, but for that case, run it on mains and then switch back to battery
The only reason I switch to mains is to make use of the power credits I have accumulated :)

Other case use would be for variable 'free hours' various wholesalers offer here. (You can select on a daily basis which hour is free)
I currently get a fixed 3 hrs free between 9pm and 12, so it doesnt really effect mine. Thats the fixed battery charge time.

jt1971, 2023-05-16:
> *Havent installed yet Before I dive in, is there a documented way to set discharge / charge times from the interface via code example? (Home Assistant) Im running a SMILE5 and the documentation and [...]

My code is scrappy but works well using timed automations to send the mqttt messages using the brilliantly engineered alpha2mqtt work.  Have got it to the point that it works semi-autonomously including managing charge/discharge with/without EV maximising benefit from Octopus Flux tariff.  

It is too scrappy to share but happy to share the principles I used in HA?

Spookster, 2023-05-17:
> It is too scrappy to share but happy to share the principles I used in HA?

Id be happy to see what you have done to give some ideas?

DannyBoy-PI, 2023-09-19:
My AlphaESS SMILE B3 can have 15, 30 & 45 minutes set for charge/discharge times. I have modified the Definitions.h, RegisterHandler.cpp and Alpha2MQTT.ino to include the new registry hex values for this, but the values for minutes always report zero. Any ideas as to what may be wrong (I followed the AlphaESS_Register_Parameter_List.pdf and ensured I have set the correct values (0x085E, 0x085F, 0x0860 & 0x0861)) ? Great project and far more reliable than trying to pull data from the Alpha Cloud.
[image]
[image]
[image]
[image]
[image]

spiderdijon, 2024-04-09:
I have also noticed 0x085E, 0x085F, 0x0860 & 0x0861 appear to be always 0 even when set to 15,30,45 minutes in the app. Has anyone managed to set the minute registers successfully?

DannyBoy-PI, 2024-04-09:
No, I haven't been able to get this to work. I contacted AlphaESS and they "claim" it is not supported on the B3. I set the times using the webpage and disconnected my B3 from the net and it charged at the specified time (10:15 -> 10:30 I seem to recall), so it does action the request locally (not a push from AlphaESS), I just have not found out where it is stored in the registry of the B3.....

dxoverdy, 2024-04-09:
They were the first registers I wrote to as a test as they were pretty safe.
I found you could burn any time you wanted:

First Start Time Charge -- 00:30
{"registerAddress": "0x0856","value":30,"dataBytes":2}
First Stop Time Charge -- 04:30
{"registerAddress": "0x0857","value":270,"dataBytes":2}
Second Start Time Charge -- 17:51
{"registerAddress": "0x0858","value":1071,"dataBytes":2}
Second Stop Time Charge -- 18:42
{"registerAddress": "0x0859","value":1122,"dataBytes":2}

Just multiply 60 by the number of hours and add on the remaining minutes.  17:51 and 18:42 will appear on the web site, but I couldn't tell you if the inverter honours is.  Sticking to 15 minutes at the very least will.

DannyBoy-PI, 2024-04-10:
Interesting, I tried to set it for 02:15 (60 x 2 + 15) = 135 using the following:-
mosquitto_pub -h <servername> -p 1883 -t Alpha2MQTT/request/write/register/raw/data -m '{"registerAddress":"0x0856","dataBytes":2,"value":135}'
And it has set the charge start time for 15:00 ? Am I being a bit dim ?

dxoverdy, 2024-04-10:
> mosquitto_pub -h -p 1883 -t Alpha2MQTT/request/write/register/raw/data -m

That’s weird.  That suggests every 1 is worth 6.666 minutes.

What does it change to if you just burn the value 1?

Also, don’t forget, with your alpha2mqtt you can ping a four hour charge
message manually at 2:15 if you are seeking to make use of a car tariff.

Regards
Dan

DannyBoy-PI, 2024-04-10:
Hi Dan,
Sending value 1 sets it to 01:00.
As for the reason behind me wanting this: On Intelligent Octopus Go, you get cheap rates during the day (if charging your car) if the weather is windy enough. I'm knocking up code to detect this and then making use of it to charge my Alpha Batts (and change the unit rate in my app) if I need too...
Cheers,
Dan M.

DannyBoy-PI, 2024-04-10:
Even weirder, setting it to 130 changes the start time to 10:00 ????

dxoverdy, 2024-04-10:
I think that makes sense.  If you work on the assumption that 1 is an hour,
5 days is 120 hours and you’re back at midnight.  The remaining 10 gets you
to ten am.

How about 0.25?

DannyBoy-PI, 2024-04-10:
sending 0.25 gives me a start time of 01:00

DannyBoy-PI, 2024-04-10:
Interesting, sending 0.50 gives me a start time of 02:00

DannyBoy-PI, 2024-04-10:
and 0.75 gives me 03:00, so I tried 0.80 and I got 08:00 !!, so I tried 1.25 and got 05:00..........

DannyBoy-PI, 2024-05-16:
Hi Dan,
Thanks for the above, but that is what I was originally trying to do before my first post to this group. I will try again though and see if it makes any difference...

DannyBoy-PI, 2024-05-16:
Hi again Dan,
Just tried the following:-

mosquitto_pub -h <server_name> -p 1883 -t Alpha2MQTT/request/write/register/raw/data -m '{"registerAddress":"0x085E","dataBytes":2,"value":25}'

And the value for minutes remains unchanged on the website/app - Do you have a legacy Smile B3 and have you tried the same ?

Cheers,

Dan M.

dxoverdy, 2024-05-16:
I have a B3 basic and Alpha told me it would never support any new features
such as the more granular time, or the discharge to grid via app, or modbus
TCP.

They’ve moved on from it unfortunately, harsh seeing as I dropped 4k on an
11kWh system but times move on.

DannyBoy-PI, 2024-05-16:
Hi Dan,

Yeah, it's odd that it supports it via the app/web (i.e. 00,15,30,45 minutes), yet it doesn't use those registers. I've proved it's not a push at the specific time from Alpha as I disconnected my system from the net and it started charging at the time I set.. So... It must be storing it somewhere in the registry, I just don't know where. I was considering knocking up a bit of code to collect all the registry data, then make the change (to say 02:45), collect the all the registry data again, and then check for differences...I will give this a go at some point, but busy with other things right now...

Thanks anyway, your code/hardware does just about everything else I need and is far superior to API's.

It is a shame that Alpha can't be bothered to support the legacy systems that have at least 4 or 5 years of life left to them. It would certainly make me think before purchasing a new Alpha system in the future, which is a shame as the hardware is good !

Dan M.

dxoverdy, 2024-05-16:
You are lucky, my web interface is still 15 min intervals and no discharge
option.

That said, don’t need that shit anyway with Alpha2MQTT

Also, don’t forget, you can use Alpha2MQTT to charge/discharge at any time
you like and for as long as you like, replicating the web interface.

But I will still keep trying with the web interface.
Dan

msegec, 2026-10-16:
Late, but for anyone with a B3 who hit the same wall: the period and SOC registers 0x0850 to 0x0859 only take a write while the matching bit of 0x084F is set. The minute registers 0x085A to 0x0861 mentioned above are untested, the same gate may apply.

The inverter acknowledges every write to the settings block 0x084F to 0x0859 with a normal function 0x10 echo, but only keeps the value when the matching bit of the time period control flag (0x084F, decimal 2127) is set at the moment of the write:

- bit 1 (value 2, discharge time control) unlocks 0x0850 to 0x0854 (UPS reserve / discharge cutoff SOC and the discharge periods),
- bit 0 (value 1, charge time control) unlocks 0x0855 to 0x0859 (charge cutoff SOC and the charge periods).

The flag itself always writes, and a value written while the bit was set stays after the bit is cleared. So the sequence is: read 0x084F, write it back with the needed bit set, write the setting, write 0x084F back to the old value, read the setting back. With the flag at 0 the same writes are acknowledged and dropped. Function 0x06 gets no reply at all on this unit.

Checked by reading each register at 1 s and 5 s after the write (SMILE-B3, EMS 1.03.78, Modbus RTU, no cloud link):

| flag at write | 0x0850 = 15 | 0x0851 = 19 | 0x0855 = 86 |
|---|---|---|---|
| 0 | kept 20 | kept 18 | kept 85 |
| 1 | kept 20 | not tried | took 86 |
| 2 | took 15 | took 19 | not tried |

Caveat: this was found at 4 am today and the battery has not yet discharged through the old 20 % cutoff, so whether the inverter honours the new value in operation is still untested. The register persistence above is what has been checked. The dispatch block 0x0880 to 0x0887 needs none of this. A write up with the exact Alpha2MQTT topics is in the README of https://github.com/msegec/Alpha2MQTT-PiPicoW.