dxoverdy/Alpha2MQTT #9: Home Assistant Errors

mercurystorm, 2023-05-18:
For the past few days a few of my displays have shown "Unknown"

A bit of googling it seems something changed / broke with 2023.5 - I just found your updated configuration.yml and applied id.. but still getting errors

May need to remove state_class from the 24 entities with enum?

Entity sensor.alpha_read_raw_register_register_address (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.alpha_read_raw_register_raw_data (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.alpha_write_raw_data_register_response_status (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.alpha_write_raw_data_register_register_address (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.alpha_write_raw_data_register_raw_data (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('enum') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

Spookster, 2023-05-24:
@mercurystorm 
did you manage to get the right format for this?

mercurystorm, 2023-05-27:
I do mine slightly different as I dont replace the configuration file - in my configuration file I have:

homeassistant:
    packages:
      pack_1: !include alphaess.yaml

and in my alphaess.yaml i have:

#MQTT
mqtt:
  button:
# Inverter Buttons
    - unique_id: alpha_set_normal
      command_topic: "Alpha2MQTT/request/set/normal"
      name: "Alpha Set Normal"
      command_template: "{\"void\": 0}"
      payload_press: "true"
      qos: 0
      retain: false
      entity_category: "config"
      icon: "mdi:lightbulb-auto"

    - unique_id: alpha_set_charge
      command_topic: "Alpha2MQTT/request/set/charge"
      name: "Alpha Set Charge"
      command_template: "{\"watts\": {{ states('input_number.input_dispatch_charge_watts')|int }},\"socPercent\": {{ states('input_number.input_dispatch_charge_soc')|int }},\"duration\": {{ states('input_number.input_dispatch_charge_time')|int }} }"
      qos: 0
      retain: false
      entity_category: "config"
      icon: "mdi:battery-positive"

    - unique_id: alpha_set_discharge
      command_topic: "Alpha2MQTT/request/set/discharge"
      name: "Alpha Set Discharge"
      command_template: "{\"watts\": {{ states('input_number.input_dispatch_discharge_watts')|int }},\"socPercent\": {{ states('input_number.input_dispatch_discharge_soc')|int }},\"duration\": {{ states('input_number.input_dispatch_discharge_time')|int }} }"
      qos: 0
      retain: false
      entity_category: "config"
      icon: "mdi:battery-negative"

    - unique_id: alpha_read_handled_register
      command_topic: "Alpha2MQTT/request/read/register/handled"
      name: "Alpha Read Handled Register"
      command_template: "{\"registerAddress\": \"{{ states('input_text.input_read_handled_register_register_address') }}\" }"
      qos: 0
      retain: false
      entity_category: "config"
      icon: "mdi:database-export"

    - unique_id: alpha_read_raw_register
      command_topic: "Alpha2MQTT/request/read/register/raw"
      name: "Alpha Read Raw Register"
      command_template: "{\"registerAddress\": \"{{ states('input_text.input_read_raw_register_register_address') }}\", dataBytes: {{ states('input_number.input_read_raw_register_data_bytes')|int }} }"
      qos: 0
      retain: false
      entity_category: "config"
      icon: "mdi:database-export"

    - unique_id: alpha_write_raw_data_register
      command_topic: "Alpha2MQTT/request/write/register/raw/data"
      name: "Alpha Write Raw Data Register"
      command_template: "{\"registerAddress\": \"{{ states('input_text.input_write_raw_data_register_register_address') }}\", dataBytes: {{ states('input_number.input_write_raw_data_register_data_bytes')|int }}, value: {{ states('input_number.input_write_raw_data_register_value')|int }} }"
      qos: 0
      retain: false
      entity_category: "config"
      icon: "mdi:database-export"

#MQTT Sensors
  sensor:
#Inverter Sensors
[... 829 lines cut ...]
    min: 6
    max: 6
    mode: text
    icon: "mdi:map-marker"
    pattern: "0x[A-Z0-9][A-Z0-9][A-Z0-9][A-Z0-9]"

  input_write_raw_data_register_register_address:
    name: "Input Write Raw Data Register Address"
    initial: "0x0880"
    min: 6
    max: 6
    mode: text
    icon: "mdi:map-marker"
    pattern: "0x[A-Z0-9][A-Z0-9][A-Z0-9][A-Z0-9]"

#Recorder
# Pointless keeping some values over time
# So looking to keep those we want to chart at any sort of frequency
# Battery charging/discharging, SOC, importing/exporting, load and PV generation
# And also the utility meters as they are supposedly going to keep us aggregate figures by day

#recorder:
#  exclude:
#    entities:
#      - sensor.alpha_battery_current

Spookster, 2023-05-27:
Well - Ive learnt something new, I didnt know you could run 'packages'
Ive split mine too, but using standard includes:

```
mqtt: !include mqtt.yaml
automation: !include automations.yaml
script: !include scripts.yaml
template: !include templates.yaml
sensor: !include sensors.yaml
scene: !include scenes.yaml
group: !include groups.yaml
utility_meter: !include utility.yaml
```

The yaml changes you made to the code fixed your errors?

dxoverdy, 2023-10-01:
I am going to assume state_class: “none” and the updated yaml provided in the project got you sorted.

closed 2023-10-01 by dxoverdy