dxoverdy/Alpha2MQTT #11: Feature Request: ESPHome support

dan-s-github, 2023-05-25:
I wonder if it would be possible to use ESPHome to build and install this package on the Controller.

Would make installation of the software much easier for some users

dan-s-github, 2023-05-28:
Just stumbled over this project here which has an interesting custom pcb for interaction with CAN
https://github.com/vekexasia/comfoair-esp32

dxoverdy, 2023-10-01:
That would be a separate project with different hardware.  If memory serves there are modbus integrations right out the box in esp home but you’d need to code and appreciate handle/calculate each register individually.

closed 2023-10-01 by dxoverdy

dan-s-github, 2023-10-02:
I ordered a https://docs.m5stack.com/en/atom/atomic485 and will do some testing on this as soon as I get access to the modbus interface. Unfortunately I currently have a replacement Smile5 v1 where I can't get access to the ports from the front.
Initial request though was to be able to build and deploy the current code using ESPHome, but that might still be tricky and not all components you use would be supported.
I am also playing with an https://shop.m5stack.com/products/atoms3-dev-kit-w-0-85-inch-screen as part of my prep