dxoverdy/Alpha2MQTT #12: Export / Import value

Spookster, 2023-05-28:
It seems to me the logic for import  / export of REG_GRID_METER_R_TOTAL_ACTIVE_POWER_1 is reversed? It may just be the way my system is reading the values?

In node red, 10 second, the code is 

`msg.payload.GRID_STATUS = msg.payload.REG_GRID_METER_R_TOTAL_ACTIVE_POWER_1 == 0 ? "Idle" : msg.payload.REG_GRID_METER_R_TOTAL_ACTIVE_POWER_1 > 0 ? "Exporting" : "Importing";`

Ive changed mine to:

`msg.payload.GRID_STATUS = msg.payload.REG_GRID_METER_R_TOTAL_ACTIVE_POWER_1 == 0 ? "Idle" : msg.payload.REG_GRID_METER_R_TOTAL_ACTIVE_POWER_1 > 0 ? "Importing" : "Exporting";`

Another minor issue, which may be specific to the SMILE5, is that REG_INVERTER_HOME_R_INVERTER_TEMP seems to be out by a factor of 10. ie , its reading a raw value of 3.5 degrees, when a more likely value is 35 degrees.

Ive changed that in nod red to use JSON msg.payload.REG_INVERTER_HOME_R_INVERTER_TEMP * 10

dxoverdy, 2023-05-28:
There is an option called legacy values.  Check it out in the
documentation.  Is your voltage out too?

Spookster, 2023-05-28:
I guess without knowing what those values should be, Im unsure?
Does this flag any issues?

[image]

dxoverdy, 2024-04-29:
Closing as stale.  If any issue persists, please re-open or re-log and I will investigate further.
With regards
Dan

closed 2024-04-29 by dxoverdy