dxoverdy/Alpha2MQTT #13: RS485 types

Spookster, 2023-06-01:
Would a smaller 3485 module such as [this ](https://www.aliexpress.com/item/1005001579669576.html?spm=a2g0o.detail.1000014.3.5ba93f40QcRJUE&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40000.326746.0&scm_id=1007.40000.326746.0&scm-url=1007.40000.326746.0&pvid=d57aa898-9240-48a0-b25d-ac516722e698&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40000.326746.0,pvid:d57aa898-9240-48a0-b25d-ac516722e698,tpp_buckets:668%232846%238109%23206&pdp_npi=3%40dis%21NZD%211.3%211.14%21%21%21%21%21%402101efeb16855787532197865e5207%2112000016669641051%21rec%21NZ%215879163) work?

Im thinking purely from a smaller form factor
[image]

dxoverdy, 2023-06-01:
I see no reason why not, go for it.

dc42, 2023-07-08:
You will need to connect the EN pin of the board to pin D5 of the ESP. Other than that, I think it should work.

The reason is that the board you pictured really does use a MAX3485 chip, whereas the larger red PCBs pictured in this project don't! They use a a MAX13487E chip. That chip has an auto transmit/receive mechanism built into it, whereas the MAX3485 doesn't. Incidentally, the MAX13487 chip is intended for 5V operation, not 3.3V; so ideally anyone using the red board should power it from 5V and use a potential divider between the TXD pin on the red board (which is the receiver output, so should really be called RXD IMO) and the ESP8266 pin to reduce the signal level to 3.3V.

Spookster, 2023-07-08:
Good information thanks!
Yes, EN to D5 does indeed work

Spookster, 2023-07-09:
What I ended up with for now. I had to make do with the only rj45 that was available

[image]

[image]

[image]

closed 2023-07-11 by Spookster