dxoverdy/Alpha2MQTT #16: Feature Request: modbus/tcp support

aw1604, 2023-07-25:
Hello,
Would it be possible to add direct modbus/tcp support ? For example, the Smile-T10 with current software update supports modbus on port 502/tcp (and I can pull data using home-assistant modbus/tcp directly). So basically I'm just missing the AlphaESS-Frontend, while the data is already in HA without any ESP/serial-to-network controller.

Best Regards,
Andreas

PS: I am not a coder, but willing to sponsor some work on this

dxoverdy, 2023-07-25:
I converted 80% into Python as I would rather use modbus over tcp but at
the last minute Alpha told me the Smile B3 doesn’t support it so I can’t
develop it further, unless someone with a working tcp ip device could give
me access to their home assistant and network details of their alpha system
so I can finish it.

There is also less interest to fight too seeing as I can’t benefit from
what I do ha!

dc42, 2023-07-26:
@aw1604 I didn't realise that Alpha ESS had implemented Modbus TCP yet, the last I heard was that it was planned. What settings in configuration.yaml are you using to access your inverter? I just tried the following but it isn't working:

```
modbus:
  - name: SMILE5
    type: tcp
    host: 192.168.1.70
    port: 502
    sensors:
      - name: "Grid power"
        slave: 0x55
        address: 0x0021
        input_type: holding
        unit_of_measurement: W
        state_class: measurement
        count: 2
        data_type: int32
        scan_interval: 10
        unique_id: "modbustcp_grid_power"
```

dxoverdy, 2023-07-26:
Very few devices support it and I get the feeling from speaking to Alpha it
is not something they have an interest in adding to existing devices.

Hope that changes of course, it would be cool!

PS needs to be on Ethernet, WiFi full stop doesn’t expose it on supported
devices.

dc42, 2023-07-26:
@dxoverdy thanks. My inverter is connected via WiFi, also it's SMILE 5 not a T10. So I'll stick with Alpha2MQTT.

RolfDeL, 2023-07-26:
Hi Dan,
If you want you can use my system. As you know the Alpha2mqtt integration is already implemented and working on my HA. Maybe this way you could double-check with the original code your TCP work.
Next week I will (hopefully) get a new internet router, so my network could change.

Also, in the near future I will get a EV charger from AlphaESS, and for now it uses the modbus.

KR 
Rolf

dxoverdy, 2024-01-05:
I'd be interested in pursuing this Rolf if you are.  Can I presume you integrated your Alpha2MQTT with something like Home Assistant?

The best way would be a Home Assistant instance opened up to the outside world so I can access.  We will install the Python integration and I'll see where I get to with the script I concocted.

Feel free to check back in on email if you fancy it.  Thanks :)