dxoverdy/Alpha2MQTT #17: Data freezing

Spookster, 2023-09-27:
I dont believe this is a software issue, but has anyone experienced a freezing of data output recently?
As below, the unit stops sending data.
[image]

Its happened multiple times over the last week, after running reliably for months.
I thought it may have been hardware, however swapping to a complete new unit has the same issue.
Network cable to unit replaced, same issue.

It makes me wonder if theres perhaps been a software update or fault at the ESS end.

Next step is to try and run #debug to see if anything shows up there

DannyBoy-PI, 2023-09-27:
Hi Dan,

No, I haven't seen that issue. Only issue I currently have is the registers for minutes (charging) don't seem to be right (for Smile B3 anyway). I have logged a ticket with Alpha to see if they can throw any light on it.

Regards,

Dan.

[image]<https://user-images.githubusercontent.com/982535/271092323-954e3b96-b68b-4a81-badc-4bbdf91bf056.png>

Spookster, 2023-10-02:
So far, resetting the main ESS unit appears to have fixed the problem

closed 2023-10-03 by Spookster

rodgers86, 2024-04-14:
Hi have you had any further issues like this? 

In the last week this has happened 3 times. 

Rebooting Node-Red in Proxmox seems to instantly fix it. But I’ve still not worked out why it’s happening's.