dxoverdy/Alpha2MQTT #18: Inconsistent retrieval and lots of timeouts

ffrog8, 2023-10-04:
I've spent a long while getting Alpha2MQTT to talk to my smile 5. After re-reading  #14  addressing some broken cable I finally got messages appearing in MQTT :). Unfortunately when reviewing the 10sec MQTT history it looks like a random set of the registers are being returned, sometimes none at all. Any suggestions, ideas welcomed.

```
10/02/2023 12:20:59 PM(-13.9 seconds)
 

{
}

10/02/2023 12:20:43 PM(-16.54 seconds)
 

{
    "REG_BATTERY_HOME_R_VOLTAGE": 50.30,
    "REG_INVERTER_HOME_R_CURRENT_L1": 3.10,
    "REG_INVERTER_HOME_R_PV5_POWER_1": 0,
}

10/02/2023 12:20:29 PM(-13.62 seconds)
 

{
    "REG_INVERTER_HOME_R_PV2_POWER_1": 2234,
    "REG_INVERTER_HOME_R_PV3_POWER_1": 0,
    "REG_INVERTER_HOME_R_PV5_POWER_1": 0,
    "REG_INVERTER_HOME_R_PV6_POWER_1": 0,
}

10/02/2023 12:20:15 PM(-14.7 seconds)
 

{
    "REG_BATTERY_HOME_R_VOLTAGE": 50.30,
    "REG_GRID_METER_R_TOTAL_ACTIVE_POWER_1": 0,
    "REG_INVERTER_HOME_R_PV5_POWER_1": 0,
    "REG_INVERTER_HOME_R_PV6_POWER_1": 0,
}

```

Here is what I see on the serial port. The Alpha Serial Number is correct but i'm not sure what to try to fix the timeouts?

```Connecting to XXXXXXX
WiFi connected, IP isXXXXXXAbout to request buffer
Requesting a buffer of : 4608 bytes
_bufferSize: 4608,

_maxPayload (Including null terminator): 4096
About To Try: 115200
Alpha Serial Number: ALXXXXXXX
Attempting MQTT connection...Connected MQTT
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
noResponse
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
Failed to addStateInfo for: 258, Result was: 4
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
Failed to addStateInfo for: 65533, Result was: 4
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
Failed to addStateInfo for: 2182, Result was: 4
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
Failed to addStateInfo for: 2183, Result was: 4
Failed to addStateInfo for: 1059, Result was: 2
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
Failed to addStateInfo for: 1063, Result was: 4
Timeout waiting for RS485 response.  Likely no more data coming.
```

dxoverdy, 2023-10-04:
I asked if you were rocking 115200 on email, maybe you missed it!  It seems
you very much are.

Try this, this worked for another fella:

“Lastly, it could be that the inverter is struggling to process lots of
messages in sequence.  Therefore, you could put a delay between each send
by adding a delay(80); to line 105 of RS485Handler.cpp (currently blank)
which will enforce a rough 1/12th second delay before requests are sent.”

ffrog8, 2023-10-04:
Okay I'll flagpole that and report back. Cheers!

ffrog8, 2023-10-04:
Mostly the same results unfortunately.

For some reason on the third reboot (output below) no Alpha serial was output?

```

Timed Out (resp->dataSize): 0
Failed to addStateInfo for: 1063, Result was: 4
▒HpI6▒▒▒▒yɨ▒▒x`▒^E▒KG▒|▒$Connecting to XXXXXX
WiFi connected, IP isXXXXXXAbout to request buffer
Requesting a buffer of : 4608 bytes
_bufferSize: 4608,

_maxPayload (Including null terminator): 4096
About To Try: 115200
Timed Out (inByteNumZeroIndexed): 7
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 2
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
Attempting MQTT connection...Connected MQTT
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
noResponse
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
Failed to addStateInfo for: 258, Result was: 4
Timed Out (inByteNumZeroIndexed): 7
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 2
Timed Out (inByteNumZeroIndexed): 7
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 2
Timed Out (inByteNumZeroIndexed): 7
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 2
Timed Out (inByteNumZeroIndexed): 7
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 2
Timed Out (inByteNumZeroIndexed): 9
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 4
Timed Out (inByteNumZeroIndexed): 9
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 4
Timed Out (inByteNumZeroIndexed): 7
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 2
Timed Out (inByteNumZeroIndexed): 9
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 4
Timed Out (inByteNumZeroIndexed): 7
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 2
Timed Out (inByteNumZeroIndexed): 9
Timed Out (gotSlaveID): 1
Timed Out (gotFunctionCode): 1
Timed Out (resp->functionCode): 3
Timed Out (gotData): 1
Timed Out (resp->dataSize): 4
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
Failed to addStateInfo for: 1077, Result was: 4
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
Failed to addStateInfo for: 65534, Result was: 4
Timeout waiting for RS485 response.  Likely no more data coming.
Timed Out (inByteNumZeroIndexed): 0
Timed Out (gotSlaveID): 0
Timed Out (gotFunctionCode): 0
Timed Out (resp->functionCode): 0
Timed Out (gotData): 0
Timed Out (resp->dataSize): 0
```

dxoverdy, 2023-10-04:
This other gent I’ve been talking to swapped over to an ESP32 which could
use hardware serial and could *possibly* result in a better experience.
 115200 is fast for the amount of data being requested in quick succession.

That said, alpha support told him he should write 0 to 0x810 in order to
swap the baud rate to 9600 which would put you in line with everyone else
and should give a more reliable connection.  Obviously Alpha2MQTT has
inbuilt functionality to do this with a single ping… but he wasn’t brave
enough.

I would swap it to 9600 but I can’t make you.  If you would like to have a
chat with the guy his email is

dxoverdy, 2023-10-04:
[image]

dxoverdy, 2023-10-04:
[image]

ffrog8, 2023-10-05:
Thanks. I'll reach out. In the mean time I tried a bit of this.

```
/request/read/register/handled
{
    "registerAddress": "0x0810",
    "dataBytes": 2
}
```

and got  (eventually after many attempts)

```
Alpha2MQTT/response/read/register/handled
{
    "responseStatus": "readDataRegisterSuccess",
    "registerAddress": "0x0810",
    "functionCode": 3,
    "registerName": "REG_SYSTEM_CONFIG_RW_MODBUS_BAUD_RATE",
    "dataType": "unsignedShort",
    "dataValue": 1,
    "formattedDataValue": "115200 (Only household)",
    "rawDataSize": 2,
    "rawData": [0,1],
    "end": "true"
}
```

Stacks up. So I figured no harm in trying to writing 1 back again?

```
/request/write/register/raw/data
{
    "registerAddress": "0x0810",
    "dataBytes": 2,
    "value": 1
}
```

Again - many many publishes later

```
Alpha2MQTT/response/write/register/raw/data
{
    "responseStatus": "writeDataRegisterSuccess",
    "registerAddress": "0x0810",
    "functionCode": 16,
    "rawDataSize": 4,
    "rawData": [8,16,0,1],
    "end": "true"
}
```

Followed by a read back

```
{
    "responseStatus": "readDataRegisterSuccess",
    "registerAddress": "0x0810",
    "functionCode": 3,
    "registerName": "REG_SYSTEM_CONFIG_RW_MODBUS_BAUD_RATE",
    "dataType": "unsignedShort",
    "dataValue": 1,
    "formattedDataValue": "115200 (Only household)",
    "rawDataSize": 2,
    "rawData": [0,1],
    "end": "true"
}
```

What i'm not sure about is how i'll know the baud rate has been changed? Does it take affect immediately? Do i need to power cycle the battery/inverter? If it does happen immediately how will i tell the difference between a noResponse that i'm getting most of the time now vs a noResponse because the baud rate has changed and Alpha2MQTT is still using the old baud rate?

dxoverdy, 2023-10-05:
/request/write/register/raw/data
{
    "registerAddress": "0x0810",
    "dataBytes": 2,
    "value": 1
}

You need to write the value 0 to swap to 9600.  Burning 1 just leaves it at
115200 as per the screenshot spec above :)

Once swapped to 9600 in theory comms will stop immediately and you’d need
to reboot your Alpha2MQTT to connect back in at 9600.  But the device may
require a reboot for it to take affect

Dan

ffrog8, 2023-10-05:
Yeah I was too chicken shit to try changing the value to 9600 on the write first attempt. Seems like I'll have to suck it and see.

dxoverdy, 2023-10-05:
LOL

ffrog8, 2023-10-06:
So i puckered up and tried

```
/request/write/register/raw/data

{
  "registerAddress": "0x0810",
  "dataBytes": 2,
  "value": 0
}
```

Multiple times because i don't actually know what to expect. I did get a reply

```
{
    "responseStatus": "writeDataRegisterSuccess",
    "registerAddress": "0x0810",
    "functionCode": 16,
    "rawDataSize": 4,
    "rawData": [8,16,0,1],
    "end": "true"
}
```

I get the impression that the raw data should be 8, 16, 0 ,0 ?
I was eventually able to read back the baud register

```
{
    "responseStatus": "readDataRegisterSuccess",
    "registerAddress": "0x0810",
    "functionCode": 3,
    "registerName": "REG_SYSTEM_CONFIG_RW_MODBUS_BAUD_RATE",
    "dataType": "unsignedShort",
    "dataValue": 1,
    "formattedDataValue": "115200 (Only household)",
    "rawDataSize": 2,
    "rawData": [0,1],
    "end": "true"
}
```

I rebooted the Alpha2MQTT - still reported 115200 as baud rate
I rebooted inverter Alpha2MQTT again reported 115200 as baud rate 

So i'm kinda stummped again?

ffrog8, 2023-10-06:
Good ideas, I'll give them a try.

ffrog8, 2023-10-06:
So I tried the force charge/discharge write. No joy. Then sitting back and thinking for a bit started wondering about library versions. I build things a little differently. TL;DR turns out I'm using esp8266 core 3.1.2 instead of 3.0.2. :facepalm: Now it works great* at the 115200 baud rate!

I use a docker container and arduino-cli to build because reasons, I didn't want to install another ide, I have two alphas and need two firmware's with different device names.

the build script looks like this.
the entrypoint tail -f is just to keep the container running so multiple builds can be done without having to reinstall all the libs into the container each time

```
#!/bin/bash 
#  export GIT_BASE=/volume1/docker_cached/apps/vscode/git/Alpha2MQTT
#  docker run -d --name arduino-cli-build -v  ${GIT_BASE}:/project --entrypoint tail solarbotics/arduino-cli:0.25.0-python3.7 -f /dev/null
#  docker exec -it arduino-cli-build /project/Alpha2MQTT/build.sh

arduino-cli core update-index --additional-urls https://arduino.esp8266.com/stable/package_esp8266com_index.json
arduino-cli core install esp8266:esp8266@3.0.2  --additional-urls https://arduino.esp8266.com/stable/package_esp8266com_index.json
arduino-cli lib install "Adafruit BusIO"
arduino-cli lib install "Adafruit SSD1306"
arduino-cli lib install "Adafruit GFX Library"
arduino-cli lib install PubSubClient

cd /project
for SER in 01 02
do
    echo "Building for ${SER}"
    arduino-cli  compile -e --build-property build.extra_flags="-DESP8266  -DDEVICE_NAME=\"Alpha2MQTT-${SER}\"" --fqbn esp8266:esp8266:nodemcuv2 Alpha2MQTT
    mv Alpha2MQTT/build/esp8266.esp8266.nodemcuv2/Alpha2MQTT.ino.bin ./Alpha2MQTT-${SER}.bin
done
```

producing an output like this
```
Building for 01
Executable segment sizes:
ICACHE : 32768           - flash instruction cache
IROM   : 314144          - code in flash         (default or ICACHE_FLASH_ATTR)
IRAM   : 29477   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...)
DATA   : 1512  )         - initialized variables (global, static) in RAM/HEAP
RODATA : 16404 ) / 81920 - constants             (global, static) in RAM/HEAP
BSS    : 26848 )         - zeroed variables      (global, static) in RAM/HEAP
Sketch uses 361537 bytes (34%) of program storage space. Maximum is 1044464 bytes.
Global variables use 44764 bytes (54%) of dynamic memory, leaving 37156 bytes for local variables. Maximum is 81920 bytes.

Used library         Version Path
SoftwareSerial       6.12.7  /home/circleci/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/SoftwareSerial
ESP8266WiFi          1.0     /home/circleci/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi
PubSubClient         2.8     /home/circleci/Arduino/libraries/PubSubClient
SPI                  1.0     /home/circleci/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/SPI
Wire                 1.0     /home/circleci/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/Wire
Adafruit_GFX_Library 1.11.7  /home/circleci/Arduino/libraries/Adafruit_GFX_Library
Adafruit_BusIO       1.14.1  /home/circleci/Arduino/libraries/Adafruit_BusIO
Adafruit_SSD1306     2.5.7   /home/circleci/Arduino/libraries/Adafruit_SSD1306

Used platform   Version Path
esp8266:esp8266 3.0.2   /home/circleci/.arduino15/packages/esp8266/hardware/esp8266/3.0.2
```

\* sometimes the REG_CUSTOM load & solar are missing from the 10sec state.

Spookster, 2023-10-06:
It seems to be a similar issue to what I experienced. Im curious as to why restarting the esp reestablishes the data connection.
Is there a handshake involved? Why would the data not restart?

```
14:57:01.349 -> Timed Out (gotData): 0
14:57:01.349 -> Timed Out (resp->dataSize): 0
14:57:01.395 -> Rx: Nothing
14:57:01.395 -> noResponse
14:57:04.769 -> Tx: 55 03 01 02 00 01 29 E2
14:57:05.185 -> Timeout waiting for RS485 response.  Likely no more data coming.
14:57:05.252 -> Timed Out (inByteNumZeroIndexed): 0
14:57:05.285 -> Timed Out (gotSlaveID): 0
14:57:05.318 -> Timed Out (gotFunctionCode): 0
14:57:05.352 -> Timed Out (resp->functionCode): 0
14:57:05.385 -> Timed Out (gotData): 0
14:57:05.385 -> Timed Out (resp->dataSize): 0
14:57:05.418 -> Rx: Nothing
14:57:05.451 -> Failed to addStateInfo for: 258, Result was: 4
14:57:05.493 -> Tx: 55 03 01 26 00 01 69 E9
14:57:05.818 -> Timeout waiting for RS485 response.  Likely no more data coming.
14:57:05.886 -> Timed Out (inByteNumZeroIndexed): 0
14:57:05.933 -> Timed Out (gotSlaveID): 0
14:57:05.966 -> Timed Out (gotFunctionCode): 0
14:57:06.000 -> Timed Out (resp->functionCode): 0
14:57:06.033 -> Timed Out (gotData): 0
14:57:06.033 -> Timed Out (resp->dataSize): 0
14:57:06.067 -> Rx: Nothing
```

ffrog8, 2023-10-06:
Restarting Alpha2MQTT ESP would put it back into its loop to find the correct baud rate?  Does yours reestablish after a restart? I recall seeing a FORCE_RESTART in the code maybe that could help.

Spookster, 2023-10-06:
During trouble shooting, I tried :

```
#define FORCE_RESTART
#define FORCE_RESTART_HOURS 1
```

Assuming it did actually force a restart ( I didnt monitor it) then no, it didnt reestablish a connection

ffrog8, 2023-10-07:
Think you've probably got more experience than I running Alpha2MQTT - I've only been up a few hours.

dxoverdy, 2024-01-05:
I had another user who rocked 115200 and got sporadic NO-RESP.  We tracked it back to his inverter needing a slight delay between requests... Call it breathing time if you will.  I've just got round to putting the functionality in the public release.

If you look at the comments in v1.26 you will see how you can enable the slight delay in Definitions.h, and also, how to change that delay to see how little a delay you can get away with.

Hope it helps you too.

dxoverdy, 2024-04-29:
Closing as stale, however please report back if the delays do not rid you of sporadic NO-RESP.
Regards
Dan

closed 2024-04-29 by dxoverdy