dxoverdy/Alpha2MQTT #19: Other hardware

vantech14, 2023-10-12:
Hi,

Is there a known "ready to use" device available that can connect via 485 into my B3?

I'd rather use a factory made device if I can. 

Contacted alpha recently and when I (eventually) got a reply, they confirm Mobus TCP isn't possible. 

Thanks.

dxoverdy, 2023-10-12:
I’m rocking a B3 too, for what it’s worth.
It’s functional but you can sorta tell Alpha have moved on and aren’t
taking older kit with them.

A bit like Samsung really.

It’s a shame as I’d have loved the opportunity to build a robust
development for modbus tcp.

Dan

vantech14, 2023-11-10:
@dxoverdy do you happen to know the Modbus slave address / ID for your b3 inverter? I have tried 85 and no luck!

Thanks.

dxoverdy, 2023-11-11:
The default of 0x55 worked for me and 100% of other folk I know of.

I’d be looking elsewhere for faults to be honest.
Regards
Dan

vantech14, 2023-11-11:
@dxoverdy I've got it working. I am however struggling to find the correct registers for the following: 

Consumption in watts (total load for the house in watts) - is this the value of the battery + grid + solar combined? 

Grid and solar total energy consumed / exported? 

Battery load (charging vs discharging) 

Any ideas / hints? Thanks.

dxoverdy, 2023-11-11:
Those are exposed in the ten second schedule

closed 2023-11-18 by vantech14

vantech14, 2023-11-18:
@dxoverdy i am trying to write the target SOC using node red. I'm running home assistant so trying to manage the full integration there for obvious reasons. 

Totally outside of this project, but would appreciate some help. 

I am trying to write a value to the target SOC register 2133 (this same register reports back the current SOC which I've proven by updating the app). It doesn't however respond to any write commands,and neither does any of the time control registers. 

Any ideas? Appreciate any input as this is driving me up the wall ;-) 

Thanks.