dxoverdy/Alpha2MQTT #20: Smile G3-S5

ettiennecoetzee, 2024-01-04:
Tried this installation on this inverter setup without success.
The inverter documentation does not include details regarding the RS485 pinout. I did what the author did and measured on all the twisted pairs to see if I can get a stable DC voltage, but couldn't find any. I hooked it up to the device using the pinout described in the notes, but no luck. Then tried all the different inverter setups, still no luck. If anybody has this specific inverter working with Alpha2MQTT, please let me know how you made it work. Thanks! Great instructions though!

dxoverdy, 2024-01-04:
Which inverter are you running and from which year?

Thanks

dxoverdy, 2024-01-04:
Ah it’s a g3-s5.  Sorry, reading on email.

Can you send a photograph of the rj45 ports?

It is likely to be 4B/5A (Blue and white B, Green A) as they all tend to
use that.

ettiennecoetzee, 2024-01-04:
My apologies, I wired the ESP32 pins incorrectly. Had a quick resolder job and its now working fine! I used the following RJ45 pinout (T568B layout):  Blue to B and Blue/white to A. Configured as the Smile B3. Thanks for the great instructions (I just didn't read them properly :)

dxoverdy, 2024-01-05:
Amazing outcome :)  Enjoy.  Be sure to send any pics in action or share any developments you get up to!

closed 2024-01-05 by dxoverdy

ettiennecoetzee, 2024-01-06:
I know this is a closed issue, but I just wanted to know if I can draw 5V DC from anywhere obvious in the inverter setup to power this nifty device?

dxoverdy, 2024-01-06:
I’ve been think about this for ages and I need to investigate.  I don’t run
my own alpha2mqtt believe it or not due to this… would be great to leach a
5V source.  Or even if there is a 12V and just use a step down buck
converter to get it to 5V.

I’ll take the case off and have a look later.

ettiennecoetzee, 2024-01-06:
Just want to mention: the effort that has gone into this must have been enormous! Just the node red code pred alone must have been massive work! Many thanks for all this!