dxoverdy/Alpha2MQTT #25: Battery self discharge

spiderdijon, 2024-05-06:
I'm not sure if there is a better place to discuss this than the issues section? Under normal conditions, it seems the battery (smile B3) registers a discharge of around 36W when it's neither charging or discharging. This does seem to cause the SoC to drop gradually. I have discovered that if I write 0 to REG_BATTERY_HOME_RW_BATTERY_MOS_CONTROL the battery status changes to "idle" and the discharge drops to 0W. However, shortly after within a few seconds it returns to around 36W and REG_BATTERY_HOME_RW_BATTERY_MOS_CONTROL has been set to 1 again automatically. Has anyone had experience in using the MOS_CONTROL register to force the battery into IDLE mode? I would like to reduce the unnecessary discharge when the battery is intended to not be doing anything.

PhilipCraigAU, 2024-05-07:
Think this is normal as the battery would supply standby / monitoring power to the Battery Management System. Mine shows about 20w in idle mode (S5)[image]

PhilipCraigAU, 2024-05-07:
Join the Alpha ESS group on facebook. Probably the best place to ask questions 

https://www.facebook.com/share/zqPX5ktSyrtxRdTJ/?mibextid=K35XfP

spiderdijon, 2024-05-10:
Thanks. I unfortunately don't have a facebook account. However, to update here I can send the REG_BATTERY_HOME_RW_BATTERY_MOS_CONTROL command every 2 seconds which does keep the battery disconnected from the inverter. The power draw does drop to 0W. Once I stop spamming the command, the system reconnects everything as normal. I'm hoping to stop the battery periodically charging its self through the night as it reaches it's minimum SoC. Normally if I stop discharging at 10% at night, by the morning it's at ~3% and the logs show periodic 2500W charging spikes for several hours prior. I'm hoping to stop charging at say 5%, disconnect the battery using the MOS command (which also has the benefit of turning the fan off) and then in the morning or just prior to a scheduled charge, stop sending the MOS command and reconnect the battery. This should avoid energy being wasted. 

What I was hoping to find out originally was which mechanism was automatically reconnecting the battery. I tried disconnecting it from the internet to see if it was a command from Alpha servers, but it was not. However, spamming the command seems like a workaround. I'll test more.

dxoverdy, 2024-05-16:
I don't think anyone in the world could advocate you bamming the system MOS control.

I'll be honest, I don't fully understand your scenario.  If it's neither charging nor discharging, it must be servicing your minimal house load of some sorts.  A house is very rarely dead to the grid, even an LED on your router is going to need servicing.

It sounds to me all you need to use is the inbuilt discharge period control on AlphaCloud or the app, which disables the battery from servicing loads between specific times of the day.  Or, you can use dispatch to put the device into mode 1 for an hour and see if that does what you expect.

If you are attempting the last option, send the following commands to
Alpha2MQTT/request/write/register/raw/data

Start Dispatch:
{"registerAddress": "0x0880","value":1,"dataBytes":2}

Dispatch Time
{"registerAddress": "0x0887","value":600,"dataBytes":4}

Dispatch Mode 1 (Battery only charged via PV):
{"registerAddress": "0x0885","value":1,"dataBytes":2}

That should disable the battery for 5 minutes (600 seconds) (by saying it will only accept charge via PV, rather than servicing loads) before going back to normal usage.  If it works as expected, you can extend the number of seconds to suit.

You can come out of dispatch mode at any time by sending
{"registerAddress": "0x0880","value":0,"dataBytes":2}

dxoverdy, 2024-05-27:
Also note that the system is powered by the battery on the DC side, not the AC side, and the unit requires power to operate which may be the small draw you are seeing.  Alpha 'fiddle' this by not showing any figures in the app < 80W or so which is why you may be wondering why Alpha2MQTT shows legitimate draw and the app shows nothing.  I verified this by looking at my Growatt app early in the morning when solar had just turned on.  It was showing generation of around 50W, Alpha2MQTT was reporting similar, but the app was at zero.

You may be asking "why is it not powered from the gird" - but it's a system can work off-grid and so has to be designed such that everything can work should the grid be unavailable... and all the circuitry is essentially DC anyway.

spiderdijon, 2024-05-27:
Yes it certainly doesn't feel right to spam the MOS control. My issue is that I frequently have times where I don't want to discharge from the battery when it is charged for example when the price of my electric drops very low. When the import rate drops below my export rate it makes sense to stop the battery servicing the loads. So I command it to not discharge which works fine but I'm dropping a reasonable number of percent every hour as the system is drawing directly from the battery. I know I won't be using the battery for X hours yet there seems to not be a mode to put it into a deep sleep. Its just a slight annoyance is all. I recognise the money that would be saved is pennies