dxoverdy/Alpha2MQTT #30: Random incorrect data

mercurystorm, 2024-09-10:
I am noticing I am getting very random data occasionally which is making my home assistant stats very wrong

Here are some examples:

2024-09-10 07:37:58 - Alpha2MQTT/state/second/ten: {^M
    "REG_BATTERY_HOME_R_SOC": 85.20,^M 
    "REG_BATTERY_HOME_R_CURRENT": 13.20,^M
    "REG_INVERTER_HOME_R_CURRENT_L1": 3.00,^M
    "REG_INVERTER_HOME_R_POWER_L1_1": 414253056,^M
    "REG_DISPATCH_RW_DISPATCH_MODE": "Unknown",^M
    "REG_DISPATCH_RW_ACTIVE_POWER_1": -1509949440,^M
    "REG_DISPATCH_RW_DISPATCH_SOC": 2576980378,^M
    "REG_DISPATCH_RW_DISPATCH_TIME_1": 25000,^M
    "REG_INVERTER_HOME_R_PV1_POWER_1": 0,^M
    "REG_INVERTER_HOME_R_PV3_POWER_1": 0,^M
    "REG_INVERTER_HOME_R_PV4_POWER_1": 0,^M
    "REG_INVERTER_HOME_R_PV5_POWER_1": 0,^M
    "REG_INVERTER_HOME_R_PV6_POWER_1": 0,^M
    "REG_CUSTOM_TOTAL_SOLAR_POWER": 122868^M

2024-09-10 07:38:09 - Alpha2MQTT/state/second/ten: {^M
    "REG_BATTERY_HOME_R_SOC": 3276.80,^M
    "REG_INVERTER_HOME_R_CURRENT_L1": 3.00,^M
    "REG_INVERTER_HOME_R_POWER_L1_1": 645,^M
    "REG_INVERTER_HOME_R_INVERTER_TEMP": 3.70,^M
    "REG_DISPATCH_RW_DISPATCH_START": "Stop",^M
    "REG_DISPATCH_RW_ACTIVE_POWER_1": 32000,^M
    "REG_DISPATCH_RW_DISPATCH_TIME_1": 25000,^M
    "REG_INVERTER_HOME_R_PV1_POWER_1": 0,^M
    "REG_INVERTER_HOME_R_PV2_POWER_1": 4278190080,^M
    "REG_INVERTER_HOME_R_PV3_POWER_1": 1308622848,^M
    "REG_INVERTER_HOME_R_PV4_POWER_1": 61430,^M

2024-09-10 08:03:41 - Alpha2MQTT/state/second/ten: {^M
    "REG_INVERTER_HOME_R_CURRENT_L1": 2.40,^M
    "REG_INVERTER_HOME_R_POWER_L1_1": 461,^M
    "REG_INVERTER_HOME_R_INVERTER_TEMP": 3.70,^M
    "REG_DISPATCH_RW_DISPATCH_START": "Stop",^M
    "REG_DISPATCH_RW_DISPATCH_MODE": "Unknown",^M
    "REG_DISPATCH_RW_ACTIVE_POWER_1": 52902,^M
    "REG_DISPATCH_RW_DISPATCH_SOC": 0,^M
    "REG_DISPATCH_RW_DISPATCH_TIME_1": 25000,^M
    "REG_INVERTER_HOME_R_PV1_POWER_1": 36,^M
    "REG_INVERTER_HOME_R_PV2_POWER_1": 44,^M
    "REG_INVERTER_HOME_R_PV3_POWER_1": 0,^M
    "REG_INVERTER_HOME_R_PV4_POWER_1": 0,^M
    "REG_INVERTER_HOME_R_PV5_POWER_1": 0,^M
    "REG_INVERTER_HOME_R_PV6_POWER_1": 0,^M
    "REG_CUSTOM_TOTAL_SOLAR_POWER": 51591^M

Any ideas what may be causing this?

mercurystorm, 2024-10-01:
I was not! I had one of the original versions running, and as it didn't have any issues it was just left - But I have upgraded last week and replaced the ESP32 (incase it was that) and the data has now been flawless!

dxoverdy, 2024-10-01:
Happy days :)  I'll close the case, but I'm really glad it's working for you.