dxoverdy/Alpha2MQTT #32: Help

Celestefano, 2024-10-10:
Good morning, can someone help me to connect my solar inverter "AlphaESS Smile5" via "PUSR RS232 RS485 Modbus RTU" in Home Assistant? 1-Inverter enabled Modbus in slave mode 2- I am using canRs485 output taking the signal A+ blue and white wire, B- blue wire. I set the converter I think correctly (it's the first time I use it and I don't even know if it's the right model for me) I followed the guides and entered all the codes with a text editor by modifying the address to communicate inside HA but I receive no signal from the sensors. Many thanks to anyone who wants to help me.

dxoverdy, 2024-10-10:
Hi there,
Sorry, I can't assist you with it I'm afraid, I have no understanding on the PUSR RS232 RS485 device and it's well outside the scope of my project.

Just try all the baud rates (9600, 115200, 19200, 57600, 38400, 14400, 4800) and slave address of 0x55 (85) decimal.

Try and read a predictable value, such as state of charge.  it's an easy one to decipher and compare to your app.
Regards
Dan

dxoverdy, 2024-10-26:
Cannot progress further.

closed 2024-10-26 by dxoverdy