dxoverdy/Alpha2MQTT #34: Storion SMILE T10

dehsgr, 2024-10-19:
Hi! Can you confirm this project is working with above system? Especially writing via ModBus? I‘m struggling with it in another project and end up with a timeout writing to ModBus (FC 16/0x10).

Thanks in advance!

dxoverdy, 2024-10-26:
Yes I know of a couple of guys in Belgium who used it on a T10 to good effect.  It's all the same protocol, I'm yet to see a device which doesn't work.

closed 2024-10-26 by dxoverdy

dehsgr, 2024-10-27:
@dxoverdy thanks for this information. Are you aware of two T10 types? There is the older version (SMILE T10, I'm asking for) and the newer version (T10-HV-INV). The newer version uses ModBus TCP. The older one uses ModBus RTU.

Windowclaw, 2024-11-06:
@dxoverdy I confirm that for at least my older Smile T10, the register writing commands according to the Modbus TCP manual V2.2 do not work. The registers are not modified, and the T10 doesn't even send any answer, neither success nor error. I exclude a communication fault, since reading registers works flawlessly.
Therefore, it would be very interesting if the guys in Belgium used the newer T10 HV-INV model, or another firmware version of the "older" SMILE T10. The puzzling thing is that the Manual V2.2 lists the "old" T10 as one of the supported models.
Here are my T10 software versions:
BMU V1.53
LMU V2.39
ISO V1.57
WR V08(159)
WR ARM-VS V19(207)
EMS-SW-VS V1.01.95D
Maybe this helps also with the T10 support of your Alpha2MQTT solution
Thank you very much!

RolfDeL, 2024-11-06:
Hi, here one of the belgian guys.

Here you have my system config:

[image]

[image]

[image]

dxoverdy, 2024-11-06:
[Alpha Modbus Protocol_V1.28.pdf](https://github.com/user-attachments/files/17652869/Alpha.Modbus.Protocol_V1.28.pdf)
[AlphaESS-Handbuch_SMILET30_ModBus_RTU_V17-DE.pdf](https://github.com/user-attachments/files/17652870/AlphaESS-Handbuch_SMILET30_ModBus_RTU_V17-DE.pdf)
[AlphaESS-Handbuch_SMILET30_ModBus_RTU_V17-EN.pdf](https://github.com/user-attachments/files/17652871/AlphaESS-Handbuch_SMILET30_ModBus_RTU_V17-EN.pdf)
[AlphaESS-Handbuch_SMILET30_ModBus_RTU_V123-DE.pdf](https://github.com/user-attachments/files/17652873/AlphaESS-Handbuch_SMILET30_ModBus_RTU_V123-DE.pdf)
[AlphaESS-Handbuch_SMILET30_ModBus_RTU_V123-EN.pdf](https://github.com/user-attachments/files/17652874/AlphaESS-Handbuch_SMILET30_ModBus_RTU_V123-EN.pdf)

V1.7 is the earliest version of the protocol and addresses I've got.
Can you tell me which address you are trying to write to, with what value, and with what JSON?

Thanks

dehsgr, 2024-11-07:
@dxoverdy I'm trying to write to e.g. 0x850 ff. from v1.23 (0x712 ff. from v1.7).

dxoverdy, 2024-11-07:
You’re sending 255 right, not ff?

dehsgr, 2024-11-07:
@dxoverdy sorry for confusion. "ff." means "and following registers". For testing I would be lucky if setting 0x850/0x712 would work. But T10 (older version) always responds with a timeout.

Windowclaw, 2024-11-07:
A sample command leading to timeout on my system is the following hex:

55 06 08 56 00 16 e7 a0

consisting of:
55: device address
06: command write single register
0856: the address to write to
0016: the hour (22:00)
e7a0: CRC16 checksum

dxoverdy, 2024-11-07:
Write single register has never worked for me.  I implemented it “just
because the manual mentions it” but I wrote in the covering notes that it
does sweet FA for me.  Swap to data register