dxoverdy/Alpha2MQTT #35: Unable to write to registers

spiderdijon, 2024-12-06:
Hi,

I have been trying to set a number of registers for example:

0x0800 - REG_SYSTEM_CONFIG_RW_MAX_FEED_INTO_GRID_PERCENT (set to 0)
0x1000 - REG_SAFETY_TEST_RW_GRID_REGULATION (set to 16 / G99)

I'm aiming to set my inverter to zero export and also to G99 as it seems it is not set correctly

I get the expected response over MQTT suggesting the write was successful but the values remain unchanged. I feel I'm doing something stupid?

Reg Write:

[image]

Reg Reads:

[image]

dxoverdy, 2024-12-06:
If you are on a B3 those settings are ignored.  They will only work for the
hybrid inverters such as the Smile5 where it can enforce how much PV power
to generate…

spiderdijon, 2024-12-07:
Ah thanks. That makes sense. Interesting that when attempting to see if the web connection was overwriting the registers I blocked the inverters access to internet with my firewall. I was surprised to find that after a while the modbus was locked out until I reconnected the internet.

msegec, 2026-10-16:
Found what looks like the gate behind acknowledged but ignored writes on a Storion SMILE-B3. The symptom here (writeDataRegisterSuccess, value unchanged) is exactly what the settings block does when the flag bit is clear. I only tested 0x084F to 0x0859, so 0x0800 and 0x1000 may well be a different story, but for the SOC and timing settings this works.

The inverter acknowledges every write to the settings block 0x084F to 0x0859 with a normal function 0x10 echo, but only keeps the value when the matching bit of the time period control flag (0x084F, decimal 2127) is set at the moment of the write:

- bit 1 (value 2, discharge time control) unlocks 0x0850 to 0x0854 (UPS reserve / discharge cutoff SOC and the discharge periods),
- bit 0 (value 1, charge time control) unlocks 0x0855 to 0x0859 (charge cutoff SOC and the charge periods).

The flag itself always writes, and a value written while the bit was set stays after the bit is cleared. So the sequence is: read 0x084F, write it back with the needed bit set, write the setting, write 0x084F back to the old value, read the setting back. With the flag at 0 the same writes are acknowledged and dropped. Function 0x06 gets no reply at all on this unit.

Checked by reading each register at 1 s and 5 s after the write (SMILE-B3, EMS 1.03.78, Modbus RTU, no cloud link):

| flag at write | 0x0850 = 15 | 0x0851 = 19 | 0x0855 = 86 |
|---|---|---|---|
| 0 | kept 20 | kept 18 | kept 85 |
| 1 | kept 20 | not tried | took 86 |
| 2 | took 15 | took 19 | not tried |

Caveat: this was found at 4 am today and the battery has not yet discharged through the old 20 % cutoff, so whether the inverter honours the new value in operation is still untested. The register persistence above is what has been checked. The dispatch block 0x0880 to 0x0887 needs none of this. A write up with the exact Alpha2MQTT topics is in the README of https://github.com/msegec/Alpha2MQTT-PiPicoW.